lueschem/edi-pi #2: Debian (Armel) on the Pi Zero?

cajuncoder, 2018-05-24:
This is not an issue in that it's a bug, but I would love to be able to run straight debian (armel) on the raspberry pi Zero.

Would this be possible? What would be involved? I've read over your project and attempted configuring it to do this with no success yet. When and if I have more time I might look into this more, as I don't quite have the knowledge and understanding yet. For now, this falls somewhere between a feature request and merely opening a topic for discussion (which, as far as I'm aware, github doesn't really facilitate?).

In any case, great project! I hope this is kept going even though debian buster is gaining support for the RPI.

PS. I originally posted this on the rpi23-gen-image project, but found your repo recently and thought I would bring it up here as well. The Pi Zero is very neglected when it comes to available OS's, so it would be great to get pure debian as an option.

lueschem, 2018-05-24:
> This is not an issue in that it's a bug, but I would love to be able to run straight debian (armel) on the raspberry pi Zero.

I am just curios: Can you specify your reasons why you are interested in a pure Debian?

> Would this be possible? What would be involved?

The root file system creation should be pretty straight forward. I see two main areas where some effort is needed:
- the kernel Debian package: I honestly do not know if there is an existing package that would match the setup
- the bootloader package raspi3-firmware which does not exist for armel yet (see https://packages.debian.org/buster/raspi3-firmware)

> In any case, great project! I hope this is kept going even though debian buster is gaining support for the RPI.

Thanks!
I am looking forward to adopt Debian buster somewhere in the future. As of now I just borrow the kernel image and the raspi3-firmware package from buster.

Since the issue/feature request is valid I will keep it open. Unfortunately I do not have enough time to take care of it by myself but I am happy to integrate pull requests.

lueschem, 2019-04-12:
I am closing this issue because there was no activity since almost one year.

closed 2019-04-12 by lueschem