lueschem/edi-pi #9: Support for Raspberry Pi 4

icedaq, 2019-12-19:
Are you planing to add support for the Raspberry Pi 4 Model B or is that currently not something that you are considering?

lueschem, 2019-12-20:
I already bought the Raspberry Pi 4 and analyzed it some weeks ago (in the meantime things might have changed):

- There are considerable changes in the bootloader setup: The Raspberry Pi bootloader now loads and modifies the device tree binary. Most of the implementations just take this modified device tree and hand it over to the Linux kernel. With my robust A/B updated that I have integrated, I would prefer to be more flexible and load the device tree binary that is specific to the partition I am currently booting from.
- U-Boot is able to handle the Raspberry Pi 4 B.
- The mainline kernel was not yet (fully) supporting the Raspberry Pi 4.

However, I was able to boot a slightly tweaked arm64 edi-pi image on the Raspberry Pi 4. Therefore I am planning to add Raspberry Pi 4 support to edi-pi in the near future.

icedaq, 2019-12-20:
That sounds great! We are currently working on a project based around the Pi 4 and I would prefer to use  edi for image generation. Let me know, if I can help you with anything.

lueschem, 2019-12-26:
I have pushed preliminary Raspberry Pi 4 B support (arm64):
https://github.com/lueschem/edi-pi/tree/rpi4_support

This are the major differences compared to the Pi 3 B:
- Instead of taking a Debian kernel I compiled the Raspberry Pi kernel for arm64.
- Instead of taking the device tree binary from the kernel I take the one that gets loaded and modified by the Raspberry Pi bootloader (I do not like this approach).

Feedback is highly appreciated!
After some more testing I will merge the rpi4_support branch to master.

lueschem, 2020-01-17:
Apart from WiFi it looks like it is working.

closed 2020-01-17 by lueschem

lueschem, 2020-01-24:
@icedaq: Now also WiFi works!