lueschem/edi-pi #12: Unable to build image with latest version.

icedaq, 2020-04-25:
Hi @lueschem 

I am unable to successfully build a `pi4-buster-arm64` image using the latest version of this repository (b7fea56) and the latest version of edi (1.5.1+u1910).

`$ git clone https://github.com/lueschem/edi-pi.git
Cloning into 'edi-pi'...
remote: Enumerating objects: 244, done.
remote: Counting objects: 100% (244/244), done.
remote: Compressing objects: 100% (157/157), done.
remote: Total 1263 (delta 98), reused 199 (delta 69), pack-reused 1019
Receiving objects: 100% (1263/1263), 17.19 MiB | 3.69 MiB/s, done.
Resolving deltas: 100% (435/435), done.
$ cd edi-pi/
$ sudo edi clean pi4-buster-arm64.yml
Success: Deleted lxc container edi-tmp-705f61d1ef95634ec328.
Success: Removed pi4-buster-arm64_edicommand_lxc_publish_a201fefc from image store.
$ sudo edi -v image create pi4-buster-arm64.yml
[...]
TASK [apt_setup : Update and upgrade apt.] ***********************************************************************************************************************
 [WARNING]: Updating cache and auto-installing missing dependency: python3-apt

fatal: [edi-tmp-705f61d1ef95634ec328]: FAILED! => {"changed": false, "cmd": "apt-get install --no-install-recommends python3-apt -y -q", "msg": "E: Package 'python3-apt' has no installation candidate", "rc": 100, "stderr": "E: Package 'python3-apt' has no installation candidate\n", "stderr_lines": ["E: Package 'python3-apt' has no installation candidate"], "stdout": "Reading package lists...\nBuilding dependency tree...\nPackage python3-apt is not available, but is referred to by another package.\nThis may mean that the package is missing, has been obsoleted, or\nis only available from another source\n\n", "stdout_lines": ["Reading package lists...", "Building dependency tree...", "Package python3-apt is not available, but is referred to by another package.", "This may mean that the package is missing, has been obsoleted, or", "is only available from another source", ""]}

PLAY RECAP *******************************************************************************************************************************************************
edi-tmp-705f61d1ef95634ec328 : ok=7    changed=2    unreachable=0    failed=1    skipped=4    rescued=0    ignored=0

Error: Command '['sudo', '-u', 'pascal', 'ansible-playbook', '--connection', 'lxd', '--inventory', '/tmp/edi-pi/tmpfv5lmso8/inventory', '--extra-vars', '@/tmp/edi-pi/tmpfv5lmso8/extra_vars_100_base_system', '/usr/lib/python3/dist-packages/edi/plugins/playbooks/debian/base_system/main.yml']' returned non-zero exit status 2.
For more information increase the log level.
`

I tried to use a different debian mirror, which resulted in the same error. I also added the package in question (`python3-apt`) to the `additional_packages` in the edi config, in which case the package gets installed fine but the image creation process still fails at the same step. I also tried to add a command task doing a `apt update` before the first apt task but this also did not help.

Do you have any idea what the problem could be and how to solve it?

Thanks,
Pascal

lueschem, 2020-04-25:
Many thanks for this detailed bug report!

I am pretty sure that the (temporary) container does not get an IPv4 address.

This is due to a setup change on the LXD side. The UDP checksum does not get calculated anymore and this confuses the isc-dhcp-client within the container.

Can you try the following workaround (on the host system):
```
sudo apt install ethtool
sudo ethtool -K lxdbr0 tx off
```
This will switch off the checksum offloading on lxdbr0. Please note that this change is not persistent and needs to be re-applied after each reboot. Funny enough, this does not happen if we have an amd64 buster container on an amd64 host. However, Debian jessie will not get an IPv4 address also within an amd64 container.

icedaq, 2020-04-26:
Hi Matthias,

Thank you very much for the quick answer! That solved the problem. I would have never figured that out.

closed 2020-04-29 by lueschem

lueschem, 2020-04-29:
I am currently preparing a new edi version that does the above solution automatically.

lueschem, 2020-04-30:
If a fix for an existing configuration is needed, please take a look at [this documentation](https://docs.get-edi.io/en/latest/upgrade_notes.html#distributions-using-nftables-instead-of-iptables) or [this commit](https://github.com/lueschem/edi-pi/commit/7f46f5b5481e5b5d7371c0a86ca71044e412bb1c).