lueschem/edi-pi #13: Display issue during u-boot execution of ARM64 RaspberryPI3 Image

UgoPoddine, 2020-06-11:
Hello,

I have used pi3-buster-arm64.yml for generating a minimal pure Debian buster arm64 image for the Raspberry Pi 3.
The result is good : the image starts as expected and i'm able to connect to the Raspberry via SSH or VNC (to LXDE) without issue.
On the contrary, during the startup process, the first startups steps are displayed on the console display (ACER AL1716) of the PI (connected to HDMI port via an HDMI2VGA converter) as usual, but after some secs of the booting process (after network initialization) an input error is shown on screen and the screen console access is no more available.
I suppose that some setting in the u-boot boot.scr are not correct, but I'm unable to troubleshoot.

Any help will be appreciated
Thank-you
Best regards
U.Poddine

lueschem, 2020-06-12:
Hi

Thanks for your detailed bug report. I just tried to reproduce the issue - but here it looks fine. I did the following:

- sudo edi -v image create pi3-buster-arm64.yml
- installed the resulting image on a Pi 3B+
- logged in as user pi
- sudo apt install lxde
- rebooted device
- successfully logged into the lxde desktop using the pi user
- switched to a Pi 3B
- repeated the successful lxde login

In both cases the monitor was connected using a HDMI-DVI adapter.

However, there are various things that could lead to a HDMI related issues. Here is a [good summary](https://howtoraspberrypi.com/raspberry-pi-hdmi-not-working/). As a first thing I would check the power supply. The wired network is usually quite power hungry and according to your report the display goes blank at around the time where the network is brought up.

Good luck
Matthias

UgoPoddine, 2020-06-12:
> Hi
> 
> Thanks for your detailed bug report. I just tried to reproduce the issue - but here it looks fine. I did the following:
> 
> * sudo edi -v image create pi3-buster-arm64.yml
> * installed the resulting image on a Pi 3B+
> * logged in as user pi
> * sudo apt install lxde
> * rebooted device
> * successfully logged into the lxde desktop using the pi user
> * switched to a Pi 3B
> * repeated the successful lxde login
> 
> In both cases the monitor was connected using a HDMI-DVI adapter.
> 
> However, there are various things that could lead to a HDMI related issue. Here is a [good summary](https://howtoraspberrypi.com/raspberry-pi-hdmi-not-working/). As a first thing I would check the power supply. The wired network is usually quite power hungry and according to your report the display goes blank at around the time where the network is brought up.
> 
> Good luck
> Matthias

UgoPoddine, 2020-06-12:
Hello Matthias,

Thank-you very much for your so fast answer.
I understand that my problem description was not so detailed as you said.

1) the raspberry is fed by a 6A max, 5V stabilized DC power supply from a lab power supply.
2) the same Raspberry, in the same configuration (same display, keyboard, sound card) runs usually with Raspbian Buster without any HW issue. The config.txt file of the Raspbian image is the standard one (no need of forcing HDMI or whatever). 
5) my description of the screen issue is not detailed : after the network initialization, where startup messages are, as usual,  shown,  the display doesn't become black, but start showing the "input not supported" error message, with the monitor led still green (the display is still in active status).

For this reason I wonder that the problem was related to some screen size setting, changing during the boot.
On your opinion, are the config.txt commands of the Raspbian bootloader also able to be used in the u-boot startup process ? In your suggested link, there are the usual config.txt instructions used by Raspbian, not the specific u-boot ones.

Thank-you 
Ugo

lueschem, 2020-06-15:
/boot/firmware/config.txt is indeed taken into account - at least to some degree. But given your additional information it could indeed be a u-boot configuration issue or a kernel issue. To narrow down the things a bit I compiled a recent Raspberry Pi kernel: https://packagecloud.io/get-edi/debian/packages/debian/buster/linux-image-5.4.42-v8+_5.4.42-v8+-1_arm64.deb
Could you try to install this kernel instead of the one that is baked into the image?

lueschem, 2020-06-15:
Maybe you want to try [this kernel](https://github.com/lueschem/edi-pi/files/4781384/test-kernel.zip).
However - to be very honest - the Raspberry Pi folks put a lot of effort into getting the GPU stuff into good shape. I usually just use the Raspberry Pi without any display attached.
To get the best GPU performance out of the Raspberry Pi your best choice is probably the Raspberry Pi OS.

UgoPoddine, 2020-06-15:
Hello Mathias,

I'm really appreciating you effort in helping me.
I will try asap :
1) your new kernel
2) the alternative one you suggested
3) some tuning in config.txt, mainly the hdmi_safe setting, hoping it can be read.

The strange is that in the first phase of boot process, the display is used correctly, and only later it will be pushed in error. Having no detailed knowledge of the booting steps it's a bit difficult for me to exactly understand where this happens.
Unfortunately I need some kind of console access to the machine... 

I will keep you informed of progresses.
Kind regards 
Ugo

UgoPoddine, 2020-06-19:
Hello Mathias,

I made the following tests :

1) I added the hdmi_safe=1 statement in the config.txt file, trying to force the minimum hdmi set-up. It got no result, still screen stops working at the middle of the startup process.
2) I have then installed successfully your patched kernel. The machine starts and runs well as before in headless mode, but again the screen went in error during the boot process. 

In any case thank-you to your support.
Cheer
Ugo

lueschem, 2020-06-22:
Hi Ugo

Many thanks for trying out the suggested fixes. Unfortunately I cannot give any further advice as I am unable to reproduce the issue here.

Best regards (and good luck in finding a solution)
Matthias

UgoPoddine, 2020-06-23:
Hi Matthias,

thank-you in any case. If I will be able to find a solution (not easy) I will keep you informed.

Cheer
Ugo

closed 2020-06-23 by UgoPoddine