lueschem/edi-pi #19: Image does not boot on CM3

icedaq, 2020-09-09:
Hi @lueschem 

I am using edi-pi in a project where we use a Raspberry Compute Module 3. I built a 64bit image without any issues (`edi -v image create edi-pi/pi3-buster-arm64.yml`) but after flashing the image, the device does not boot and is stuck on the infamous rainbow screen.

After some research I found out, that the u-boot version that is used in edi-pi (v2019.10) probably does not work on the CM3: [https://lists.denx.de/pipermail/u-boot/2020-May/411691.html](https://lists.denx.de/pipermail/u-boot/2020-May/411691.html)

The versions between 2019.07 and 2020.04 do not seem to work on the CM3. I am in the process of building the latest version of u-boot (v2020.07) using your `generate-archives` script. If everything goes well, should I open a PR to update the `u-boot-bootloader.*.tar.gz` archives or do you think that will introduce other problems/incompatibilities?

As always, thank you for your help!

Best,
Pascal

lueschem, 2020-09-10:
Hi Pascal

Support for the CM3 would be great - a PR will be highly appreciated.
I wonder if you also need a custom kernel - or does the Debian kernel already have the required DTBs?

Best regards
Matthias

icedaq, 2020-09-21:
Hi Matthias,

After switching the U-Boot version to 2020.04 the CM3 is not stuck on the rainbow screen anymore. Support for the CM3 was added to the kernel in version 4.20, which is just a bit newer than the 4.19 kernel that ships with buster. To verify that a newer kernel will work, I wrote a small ansible role that updates the kernel image to debian testing (bullseye, kernel 5.8.0-1). My playbook seems to work fine (boot.scr is updated in the system partition etc.) but when trying to boot the image I get the same message as with the 4.19 kernel:

```
[snip]
Successfully loaded secondary boot script.
## Executing script at 02400726
22421360 bytes read in 929ms (23 MiB/s)
26458574 bytes read in 1097 ms (23 MiB/s)
ERROR: Did not find a cmdline Flattened Device Tree

Starting kernel ...
```

And after that the system is just stuck. When I look at the built image, I see that I actually have a `bcm2837-rpi-cm3-io3.dtb` file in my `lib` folder so the device tree should be there but maybe it is not loaded somehow? 
Do you have any idea, what is wrong here?

Best,
Pascal

lueschem, 2020-09-21:
Hi Pascal

It might be that the DTB file name is not fully aligned between the Kernel and the U-Boot you are using.
Could you interrupt the boot-up of the CM3 and execute the following command in the U-Boot command shell:

```
printenv fdtfile
```

Note: `fdtfile` is being used within [edi-boot-shim](https://github.com/lueschem/edi-boot-shim/blob/master/template/boot.cmd.rpi.arm64).

Instead of using the Debian kernel we could also try out the Raspberry Pi OS vendor kernel. This was also required for the [Raspberry Pi 4](https://www.get-edi.io/Getting-Started-with-a-new-Embedded-System/). It should be easy to adjust the kernel build instructions for the Raspberry Pi CM3 using [this documentation](https://www.raspberrypi.org/documentation/linux/kernel/building.md).

Best regards
Matthias

P.S.: Unfortunately I have no CM3 available to test out the setup. If needed, we can also do a short screen sharing session.

icedaq, 2020-09-21:
That was a great tip! `fdtfile` gives me `fdtfile=broadcom/bcm2837-rpi-cm3.dtb` but the file provided with the kernel is called `bcm2837-rpi-cm3-io3.dtb`. After running:
```
setenv fdtfile broadcom/bcm2837-rpi-cm3-io3.dtb
boot
```
The system boots up just fine. 

But as you said, it would probably make sense to use the Raspberry Pi OS kernel. Do you have any special mechanism to switch the kernel for the Pi4? From what I can tell, you only specify a different version of the kernel and it then gets installed from the private edi-pi debian repo? Or did I miss something?

As always, thank you very much for your help!

Best regards,
Pascal

lueschem, 2020-09-21:
Hi Pascal

You are absolutely correct about the edi-pi Debian repository. The Pi 4 kernel will be fetched from [Packagecloud](https://packagecloud.io/get-edi/debian). If helpful, I can also upload your kernel Debian package there. On some other projects we also added the possibility to install a locally available kernel. Something like:

```
- name: Install kernel {{ kernel_image }} from repository.
  apt:
    name: "{{ kernel_image }}"
    state: present
    install_recommends: no
  when: not kernel_image.endswith(".deb")

- name: Copy local kernel {{ kernel_image }}.
  copy:
    src: "{{ kernel_image }}"
    dest: /tmp
  when: kernel_image.endswith(".deb")

- name: Install local kernel {{ kernel_image }}.
  apt:
    deb: "/tmp/{{ kernel_image }}"
  when: kernel_image.endswith(".deb")
```

This setup is also very helpful during development.
Of course you can also integrate your company repository (but then it will get difficult to upstream the changes to edi-pi).

Best regards
Matthias

closed 2022-11-05 by lueschem