lueschem/edi-pi #20: Updating mender-client from 1.7 to 2.4 for mender artifact format 3: missing link /var/lib/mender -> /data/mender

vw-scanomat, 2020-10-23:
I just updated my local mender-client using https://docs.mender.io/downloads by updating the board_support/roles/mender/tasks/main.yml with:
- name: Install mender-client.                                                  
  apt:                                                                          
    deb: "https://d1b0l86ne08fsf.cloudfront.net/2.4.0/dist-packages/debian/arm64/mender-client_2.4.0-1_arm64.deb"
    state: present                                                              
    install_recommends: no

This is somewhat a hack, sure.

In 1.7 there was a link at /var/lib/mender -> /data/mender, who created that?

vw-scanomat, 2020-10-23:
mender-client 1.7 used a link: /var/lib/mender -> /data/mender
mender-client 2.4 uses default /var/lib/mender, but has a parameter --data (-d) to point at a data directory such as /data/mender. Adding this to service file should fix it.

lueschem, 2020-10-23:
Indeed - the difference is between the Debian package provided by Mender and the Debian package provided by Debian. The one from Debian installs the symlink.
As a workaround the mender installation role could add this symlink. Another possibility is to take the package from Debian:
https://packages.debian.org/sid/mender-client

vw-scanomat, 2020-10-23:
I have a proposal soon.