lueschem/edi-pi #22: Image does not start on Raspberry Pi 4 8Gb

eugcel, 2020-12-03:
# Issue Type

- Bug report

# Environment detail

Raspberry Pi 4 8Gb

# Desired Behaviour

Start with no error.

# Actual Behaviour (Bug report only)
Trying to start pure image:
[image]
After  replacing start4.elf and fixup4.dat we`re getting the following output:
```
PM_RSTS: 0x00001000
RPi: BOOTLOADER release VERSION:c305221a DATE: Sep  3 2020 TIME: 13:11:46 BOOTMODE: 0x00000006 part: 0 BUILD_TIMESTAMP=1599135103 0xf82a162b 0x00d03114
uSD voltage 3.3V
Initialising SDRAM 'Micron' 32Gb x2 total-size: 64 Gbit 3200
VLI: HUB2: 0xfff00000 0x24e6 MCU: 0xfff20000 0x15218
VL805 0xfff00000 0xfff20000
XHCI-STOP
xHC ver: 256 HCS: 05000420 fc000031 00e70004 HCC: 002841eb
xHC ports 5 slots 32 intrs 4
Boot mode: SD (01) order 0
SD HOST: 250000000 CTL0: 0x00000000 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
SD HOST: 250000000 CTL0: 0x00000f00 BUS: 100000 Hz actual: 100000 HZ div: 2500 (1250) status: 0x1fff0000 delay: 1080
CID: 001b534d454331533530ddb5573c0149
CSD: 400e00325b590001dd7f7f800a400000
SD: bus-width: 4 spec: 2 SCR: 0x02058083 0x00000000
SD HOST: 250000000 CTL0: 0x00000f04 BUS: 50000000 Hz actual: 41666666 HZ div: 6 (3) status: 0x1fff0000 delay: 2
MBR: 0x00000800,  262144 type: 0x0c
MBR: 0x00040800,  524288 type: 0x83
MBR: 0x000c0800,  818854 type: 0x83
MBR: 0x00000000,       0 type: 0x00
lba: 2048 oem: 'mkfs.fat' volume: '  V       ^ '
rsc 4 fat-sectors 256 c-count 65399 c-size 4 r-dir 1 r-sec 32
PM_RSTS: 0x00001000
Partition: 0
lba: 2048 oem: 'mkfs.fat' volume: '  V       ^ '
rsc 4 fat-sectors 256 c-count 65399 c-size 4 r-dir 1 r-sec 32
Read config.txt bytes      100 hnd 0x00000036 hash '36648a4e07609ce6'
recover4.elf not found (6)
recovery.elf not found (6)
Read start4.elf bytes  2225376 hnd 0x00002f80 hash '9485ff67b88c6c1f'
Read fixup4.dat bytes     5417 hnd 0x000005e2 hash 'bf08943ed9ab6e07'
0x00d03114 0x00000000 0x0000001f
MEM GPU: 76 ARM: 948 TOTAL: 1024
Starting start4.elf @ 0xfec00200 partition 0
MMC:   emmc2@7e340000: 0, mmcnr@7e300000: 1
Loading Environment from FAT... OK
In:    serial
Out:   serial
Err:   serial
Net:   Net Initialization Skipped
No ethernet found.
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
1830 bytes read in 12 ms (148.4 KiB/s)
## Executing script at 02400000
542 bytes read in 17 ms (30.3 KiB/s)
Successfully loaded secondary boot script.
## Executing script at 02400726
libfdt fdt_check_header(): FDT_ERR_BADMAGIC
No FDT memory address configured. Please configure
the FDT address via "fdt addr <address>" command.
Aborting!
: No match
13392384 bytes read in 854 ms (15 MiB/s)
ERROR: Did not find a cmdline Flattened Device Tree

Starting kernel ...
```
and stucking on Starting kernel...

lueschem, 2020-12-03:
Many thanks for reporting this issue!

With the Raspberry Pi 4B 8GB (revision 1.4) significant modifications were done to the PCB.
As you already found out the Raspberry Pi firmware needs to be upgraded. Most likely also the U-Boot and the kernel need an upgrade.
The issue you reported above shows that the device tree cannot be successfully loaded. Subsequently the kernel fails to start.
The device tree on the Raspberry Pi 4B is treated in a very special way:

1. The Raspberry Pi firmware loads it.
2. The Raspberry Pi firmware does a lot of changes to it.
3. U-Boot takes it as is and passes it to the kernel.

Now the issue is that with the upgrade of the Raspberry Pi firmware the stuff does not fit together anymore.

I will further dig into that issue tomorrow.

lueschem, 2020-12-03:
This blog post helps with the "device_tree_address" setup:
https://waldorf.waveform.org.uk/2020/groovy-boot-modes.html

eugcel, 2020-12-03:
In the case of using U-boot, I've just found the solution to the "device tree" problem.

1. Setup Raspbian (other systems you need) onto an SD card.
2. Update and upgrade the system and the Raspberry Pi firmware (use `sudo apt full-upgrade` in the case of Raspbian)
3. With the help of Device Tree Compiler create dts file from the device-tree directory on the running system. Use the next command.
`dtc -I fs -O dts -o bcm2711-rpi-4-b.dts /sys/firmware/devicetree/base`
4. Compile dts file to dtb
`dtc -I dts -O dtb -o bcm2711-rpi-4-b.dtb bcm2711-rpi-4-b.dts`

Now you have the dtb file with all the needed changes the Raspberry firmware does.

lueschem, 2020-12-03:
Just curious: What are the changes that the Raspberry Pi firmware does to the dtb?

Are the changes depending upon the board revision and the amount of available RAM?

If so, then this would unfortunately not be a generic solution.

eugcel, 2020-12-03:
> Just curious: What are the changes that the Raspberry Pi firmware does to the dtb?

The Raspberry Pi firmware doesn't do any changes to the dtb directly. It combines the dtb with overlays in some way during the boot then passes the result to the kernel. To the best of my knowledge,  we can treat the dtb file as a some kind of base device tree and the result of the combining process as a fully resolved device tree. We need the last one to start the kernel successfully.

> Are the changes depending upon the board revision and the amount of available RAM?

Frankly, I don't know. In my estimation, it rather depends on the version of the firmware.

> If so, then this would unfortunately not be a generic solution.

I think so.

lueschem, 2020-12-03:
Also interesting:
https://www.raspberrypi.org/forums/viewtopic.php?t=276146

eugcel, 2020-12-04:
[This one](https://www.linux4sam.org/bin/view/Linux4SAM/PatchingDTinUboot) seems useful.

lueschem, 2020-12-04:
Could you check if an image built from the branch rpi4b_8gb boots on the Raspberry Pi 4B 8GB?
I was only able to verify it on a Pi 4B 4GB.

eugcel, 2020-12-04:
Unfortunately, it fails on the step of saving the environment. Here is the output.

```
U-Boot 2020.10 (Dec 02 2020 - 10:09:23 +0000)

DRAM:  7.9 GiB
RPI 4 Model B (0xd03114)
MMC:   mmcnr@7e300000: 1, emmc2@7e340000: 0
Loading Environment from FAT... *** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   eth0: ethernet@7d580000
PCIe BRCM: link up, 5.0 Gbps x1 (SSC)
starting USB...
Bus xhci_pci: probe failed, error -110
No working controllers found
Hit any key to stop autoboot:  0
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
1830 bytes read in 9 ms (198.2 KiB/s)
## Executing script at 02400000
Going to write environment on first boot!
Saving Environment to FAT... "Synchronous Abort" handler, esr 0x96000004
elr: 00000000000a2e5c lr : 00000000000a2e94 (reloc)
elr: 000000003b37be5c lr : 000000003b37be94
x0 : 000000003af66800 x1 : 0018001700160015
x2 : 000000003af54e50 x3 : 0024002300220021
x4 : 0000000000000015 x5 : 0000000000000015
x6 : 000000003b3d2f48 x7 : 000000003b3d2f48
x8 : 0000000000000001 x9 : 0000000000000008
x10: 000000003af6eef0 x11: 000000003af6eef0
x12: 0000000000000810 x13: 0000000000000001
x14: 0000000000000000 x15: 0000000000000002
x16: 0000000000004110 x17: 0000000000000000
x18: 000000003af54d90 x19: 0000000000000000
x20: 0000000000000006 x21: 0000000000000006
x22: 0000000000000000 x23: 000000003af42bd8
x24: 000000003af68ec0 x25: 000000003af42e78
x26: 000000003af43000 x27: 0000000000000000
x28: 000000003aff06e0 x29: 000000003af42b50

Code: 54000061 d2800000 d65f03c0 f9400401 (b9400821)
Resetting CPU ...

resetting ...

```

lueschem, 2020-12-04:
Ouch - looks like this: https://www.mail-archive.com/u-boot@lists.denx.de/msg385161.html

lueschem, 2020-12-04:
According to your advice I downgraded the Pi4 binary to v2020.07. It still boots on the Pi4 4GB.
Soon I should also receive an 8GB version...

eugcel, 2020-12-05:
Great job! Now it works properly!

```
U-Boot 2020.07 (Dec 04 2020 - 16:40:45 +0000)

DRAM:  7.9 GiB
RPI 4 Model B (0xd03114)
MMC:   mmcnr@7e300000: 1, emmc2@7e340000: 0
Loading Environment from FAT... OK
In:    serial
Out:   serial
Err:   serial
Net:   eth0: ethernet@7d580000
Hit any key to stop autoboot:  0
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
1830 bytes read in 12 ms (148.4 KiB/s)
## Executing script at 02400000
544 bytes read in 16 ms (33.2 KiB/s)
Successfully loaded secondary boot script.
## Executing script at 02400726
bootargs=coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:CC:5B:40 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=/dev/mmcblk0p3 rw rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
15839744 bytes read in 672 ms (22.5 MiB/s)
## Flattened Device Tree blob at 02600000
   Booting using the fdt blob at 0x2600000
   Using Device Tree in place at 0000000002600000, end 000000000260edf7

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.4.72-v8+ (lueschem@edi-pi-cross-dev-buster) (gcc version 8.3.0 (Debian 8.3.0-2)) #1 SMP PREEMPT Wed Dec 2 07:49:11 UTC 2020
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.4
[    0.000000] efi: Getting EFI parameters from FDT:
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000002c000000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] percpu: Embedded 31 pages/cpu s89304 r8192 d29480 u126976
[    0.000000] Detected PIPT I-cache on CPU0
..........

Debian GNU/Linux 10 raspberry ttyS0

raspberry login:

```

lueschem, 2020-12-18:
The 8GB Raspberry Pi 4B is finally booting.

closed 2020-12-18 by lueschem