lueschem/edi-pi #25: edi image create error for RPi4

mfaccin, 2021-12-07:
Hello everyone!

Cloning the repository and trying to create the image by running `sudo edi -v image create pi4-bullseye-arm64.yml` I got:

```
I: Extracting libxxhash0...
I: Extracting liblzma5...
I: Extracting zlib1g...
INFO:root:Running command: ['chroot', '/home/mauricio/edi-pi/tmp6ue2mb_f/rootfs', '/debootstrap/debootstrap', '--second-stage']
W: Failure trying to run:  /sbin/ldconfig
W: See //debootstrap/debootstrap.log for details
Error: Command '['chroot', '/home/mauricio/edi-pi/tmp6ue2mb_f/rootfs', '/debootstrap/debootstrap', '--second-stage']' returned non-zero exit status 1.
For more information increase the log level.
```

Any ideas on how to solve or debug it?

lueschem, 2021-12-07:
I actually faced a similar issue: https://github.com/lueschem/edi/issues/73

What is your QEMU and kernel version?

```
dpkg -l qemu-user-static
uname -a
```

mfaccin, 2021-12-07:
Thanks for the reference, will check it out.

> What is your QEMU and kernel version?

```
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name             Version           Architecture Description
+++-================-=================-============-==================================================
ii  qemu-user-static 1:4.2-3ubuntu6.18 amd64        QEMU user mode emulation binaries (static version)
```

```
Linux Lenovo-ideapad-720S-T 5.11.0-41-generic #45~20.04.1-Ubuntu SMP Wed Nov 10 10:20:10 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
```

lueschem, 2021-12-07:
Indeed - you have an old QEMU but a recent kernel. The old QEMU seems to be struggling with the new kernel. You could try to use the QEMU provided by the Ubuntu Server Backports Team: https://launchpad.net/~canonical-server/+archive/ubuntu/server-backports

A similar thing also happened on Armbian: https://forum.armbian.com/topic/16740-debootstrap-base-system-second-stage-failed/

mfaccin, 2021-12-07:
True, I updated with the QEMU from the Ubuntu Server Backports Team and it worked. Thank you @lueschem :)

closed 2021-12-07 by mfaccin