lueschem/edi-pi #31: Iproute2 is missing

Gizmo091, 2023-02-09:
Hi, 

First, thanks for your work. 

I try to create an image of debian for my raspberry pi 4, i have follow all your instruction, but i have an error when i run the command : 

```
sudo edi -v image create pi4-bullseye-arm64.yml
```
Error displayed :
```
IINFO:root:Running command: ['sudo', '-u', 'mvedie', 'ansible-playbook', '--connection', 'lxd', '--inventory', '/home/mvedie/codes/tools/raspberrypi/edi-pi/tmpm2wap31d/inventory', '--extra-vars', '@/home/mvedie/codes/tools/raspberrypi/edi-pi/tmpm2wap31d/extra_vars_100_base_system', '/usr/lib/python3/dist-packages/edi/plugins/playbooks/debian/base_system/main.yml']

PLAY [all] ****************************************************************************************************************************************************************

TASK [lxc_interfaces : Check if iproute2 is installed.] *******************************************************************************************************************
fatal: [edi-e17f2af7-3f7afe4b]: FAILED! => {"changed": false, "module_stderr": "/bin/sh: 1: /usr/bin/python: not found\n", "module_stdout": "", "msg": "MODULE FAILURE", "rc": 127}
...ignoring

TASK [lxc_interfaces : Stop if iproute2 is missing.] **********************************************************************************************************************
fatal: [edi-e17f2af7-3f7afe4b]: FAILED! => {"changed": false, "msg": "Please install iproute2 during bootstrapping (bootstrap/additional_packages)."}
	to retry, use: --limit @/usr/lib/python3/dist-packages/edi/plugins/playbooks/debian/base_system/main.retry

PLAY RECAP ****************************************************************************************************************************************************************
edi-e17f2af7-3f7afe4b      : ok=1    changed=0    unreachable=0    failed=1   

Error: Command '['sudo', '-u', 'mvedie', 'ansible-playbook', '--connection', 'lxd', '--inventory', '/home/mvedie/codes/tools/raspberrypi/edi-pi/tmpm2wap31d/inventory', '--extra-vars', '@/home/mvedie/codes/tools/raspberrypi/edi-pi/tmpm2wap31d/extra_vars_100_base_system', '/usr/lib/python3/dist-packages/edi/plugins/playbooks/debian/base_system/main.yml']' returned non-zero exit status 2.
For more information increase the log level.
```

iproute2 is installed on my machine. 
I have follow the edi installation guide and stop it at the "Building a First Container" title.

Have you an idea of the cause ? 

Thx a lot.

lueschem, 2023-02-09:
Many thanks for reporting this issue. My first guess is that the error message is misleading and we are in fact stumbling over an Ansible upstream change on the command module. I would like to reproduce this issue. Could you give me the edi version and the Ansible version you are using. Furthermore the operating system you are using might be interesting too:

```
ansible --version
edi version
```

Gizmo091, 2023-02-09:
Yes , and thx for your time : 

mvedie@vr21:~/codes/tools/raspberrypi/edi-pi$ ansible --version
ansible 2.5.1
  config file = /etc/ansible/ansible.cfg
  configured module search path = [u'/home/mvedie/.ansible/plugins/modules', u'/usr/share/ansible/plugins/modules']
  ansible python module location = /usr/lib/python2.7/dist-packages/ansible
  executable location = /usr/bin/ansible
  python version = 2.7.17 (default, Nov 28 2022, 18:51:39) [GCC 7.5.0]
mvedie@vr21:~/codes/tools/raspberrypi/edi-pi$ edi version
1.13.3+u1804
mvedie@vr21:~/codes/tools/raspberrypi/edi-pi$ cat /etc/os-release
NAME="Ubuntu"
VERSION="18.04.6 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.6 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic

lueschem, 2023-02-09:
Perfect - I was able to reproduce the issue.

Ansible 2.5.1 wants to use Python 2 for running the modules in the LXD container. However, Python 2 is not installed anymore. I can force Ansible to take Python 3 but then it fails a few tasks later. If you want to stick to Ubuntu 18.04 then you can install a more recent version of Ansible like this:

```
sudo add-apt-repository ppa:ansible/ansible
sudo apt install ansible
```

Like this I got the setup working even on Ubuntu 18.04. After all I would recommend the upgrade to Ubuntu 20.04 or even 22.04.

I will remove the Ubuntu 18.04 support from the next edi version as it is obviously not working properly anymore.
Thanks again for reporting this issue!

Gizmo091, 2023-02-23:
Hi, 

Sorry for my delayed response, i have been pushed on another task for the lasst weeks :/ 

Thx for the solution, it works fine now.

howevert my distro contains python2 
```shell
$ ls -l /usr/bin/python
lrwxrwxrwx 1 root root 9 avril 16  2018 /usr/bin/python -> python2.7
```

Thx for your time.

closed 2023-02-23 by Gizmo091

lueschem, 2023-02-23:
Many thanks for the feedback!

About the Python 2 vs Python 3: There is the host system (in your case Ubuntu) where you still have Python 2 and the target system (Debian bullseye running in the LXD container) where we do not have Python 2 anymore. Ansible runs the playbook on the host system and generates snippets of Python code that then run on the target system. Therefore we have two Python interpreters that come into play.