lueschem/edi-pi #35: Need to Extend Data Partition After Resizing RootFS

schorsche32, 2025-01-13:
Hey there,

I need a modification to the resizing script for my application. Specifically, I want the data partition (p2) to occupy the unallocated space after:
- The first rootfs partition (p3) is extended to approximately twice its original size.
- A second rootfs partition (p4) is created with the same size as p3
Here is the initial layout:

```
+------------+----------+--------+---------------------+
| p1         | p2       | p3     |  (unallocated)|
+------------+----------+--------+---------------------+
^ Boot       ^ /data    ^ RootFS
```

And here is the desired layout:

```
+-------------+-----------------------+---------------+--------+
| p1          | p2 (extended)         | p3            | p4     |
+-------------+-----------------------+---------------+--------+
^ Boot        ^ /data                 ^ RootFS_A      ^ RootFS_B
```

My initial idea was to create a fifth primary partition, but that isn’t possible due to the partition limit. Directly extending p2 also appears non-trivial, since it currently overlaps with p3.

Does anyone have a suggestion for how to solve this?

Thank you in advance for any help!

Best regards,
Georg

lueschem, 2025-01-14:
I actually like your idea with the fifth partition. AFAIK the Raspberry Pi bootloader does support an MBR/GPT hybrid partition table setup. With a GPT partition table you can go beyond 4 partitions. The Home Assistant community also shared [some obstacles they encountered with the GPT setup](https://github.com/raspberrypi/rpi-eeprom/issues/585) (along with how they removed those obstacles).

Another possibility would be to boot into an initial ramdisk and do the partition resizing and moving from there. However, I personally prefer your approach of using 5 partitions.

schorsche32, 2025-01-14:
I have now developed a new approach.

**Initial layout:**
```
+------------+--------+---------------------+
| p1         | p2    |  (unallocated)|
+------------+--------+---------------------+
^ Boot       ^ RootFS
```

**Final layout:**
```
+-------------+------------------+---------------+--------------+
| p1          | p2               | p3            |p4 (extended)
+-------------+------------------+---------------+--------------+
^ Boot         ^ RootFS_A        ^ RootFS_B      ^ /data               
```

To implement this approach, I need to:
- Create the p4 partition during the first boot.
- Copy the files from `rootfs/data` to the p4 partition.
- Mount the p4 partition to `/data`.
- During A/B updates, remove the contents of the newly downloaded rootfs in the `/data` folder.

Currently, this approach appears to be working well.

I will share it soon in a fork.

closed 2025-03-28 by lueschem